sipeed/platform-kendryte210 #2: Topics for repository

ivankravets, 2019-04-25:
It would be great if you add the next topics to this repository:
* `platformio`
* `platformio-platform`
* `risc-v`.

Developers will be able to list open source dev/platforms for PlatformIO. See
- https://github.com/topics/platformio-platform

You can do that on the main page of the repository via "Manage topics" button. See
[image]

btx000, 2019-04-25:
added

ivankravets, 2019-04-25:
Thanks! Do you plan to add more changes? Currently, if you make a new release of this dev/platform, you need to ping us somewhere to re-deploy to our global registry. We will fix that with new PlatformIO Registry where everyone will be able personally to manage own dev/platforms.

btx000, 2019-04-25:
Thanks for your reminder, we are testing on various platforms, and we can release them after the test is passed.

closed 2019-04-27 by btx000