sipeed/platform-kendryte210 #3: Rename "kendryte210-standalone-sdk" to "kendryte-standalone-sdk"

ivankravets, 2019-05-01:
It would be good to have unified name for Kendryte Standalone SDK => https://github.com/kendryte/kendryte-standalone-sdk

Could you rename it in these places?

- https://github.com/sipeed/platform-kendryte210/blob/master/platform.json#L22
- https://github.com/sipeed/platform-kendryte210/blob/master/platform.json#L23
- https://github.com/sipeed/platform-kendryte210/blob/master/platform.json#L24
- https://github.com/sipeed/platform-kendryte210/blob/master/platform.json#L41

ivankravets, 2019-05-03:
@btx000 could you add me to `Collaborators` list for this repo?

btx000, 2019-05-03:
> @btx000 could you add me to `Collaborators` list for this repo?

Added. https://github.com/sipeed/platform-kendryte210/invitations

commit 74583589 (2019-05-03): Rename "kendryte210-standalone-sdk" to "kendryte-standalone-sdk" // Resolve #3
closed 2019-05-03 by ivankravets