sipeed/platform-kendryte210 #6: Some improvements

ivankravets, 2019-05-03:
Sorry, I don't have rights to modify project links. Could I ask a few changes?

- On the home page of this repository specify **Website** to `https://platformio.org/platforms/kendryte210`
- Update **Description** to `Kendryte K210: development platform for PlatformIO`

We added extended information and docs on our side:
- https://platformio.org/platforms/kendryte210
- https://docs.platformio.org/en/latest/platforms/kendryte210.html

btx000, 2019-05-04:
You can update this information according to your ideas.

ivankravets, 2019-05-04:
I can't. Github does not allow me :( This option is available only for repository owners.
[image]

btx000, 2019-05-04:
Sorry, I just understood that there is something wrong. It has changed now.

closed 2019-05-04 by ivankravets