sipeed/platform-kendryte210 #7: 无法下载例程

hhhhljp, 2019-05-03:
[ERROR] enum34 must be installed, run `pip install enum34` 
*** [upload] Error 1
我已经安装过enum34了，是安装位置有问题还是什么其他情况吗
编译过程没有问题，例程选择的是kendryte-standalone-sdk的HelloWord

btx000, 2019-05-04:
@ivankravets 

How to set up can automatically install kflash dependency package?

ivankravets, 2019-05-04:
You need to pack kflash tool with pre-installed Python packages. See example with https://bintray.com/beta/#/platformio/dl-packages/tool-pulp-debug-bridge?tab=files.
Please download this file `tool-pulp-debug-bridge-darwin_x86` and take a look at `scripts/build.py`.
Lines: 78-107.

So, the core idea is to install dependencies to `python/site-packages` and later inject them into Python `sitedir`.

hhhhljp, 2019-05-04:
The problem has been solved, I made the download location of enum34 wrong, it should be installed in the file directory of pio

ivankravets, 2019-05-04:
This is a bug. You no need to install extra packages with PlatformIO. Package should contain all dependencies.

btx000, 2019-05-04:
I saw your example, but I still can't figure out what happened to the package during the installation process. Can I use pip directly during installation? Kflash has been released as a package in pypi, and its setup.py may be able to handle these dependencies.

ivankravets, 2019-05-05:
We don't run `python setup.py` when installing a package. We just unpack it and move to `~/.platformio/packages` folder. 
It's not easy to install Python package because it can lead to a lot of issues. There are built-in Python installation in OS, custom virtual environment for PIO Core etc... Each of them has own `site-packages` storage.

So, keeping package totally independent of `pip` resolves all issues with local Python installation. Also, it resolves issue with install package without administrative/sudo permissions.

ivankravets, 2019-05-05:
We need to pack `enum34` and `pyelftool` into the package. Could you try to do that?

btx000, 2019-05-05:
I am trying...
I noticed this function,
```python
def install_pysite(dst, packages):
    for package in packages:
        subprocess.call(["pip", "install", "--no-compile", "-t", dst, package])
```

But I don't know how it is called. Is it executed once in kflash.py every time it runs?

ivankravets, 2019-05-05:
You can run pip install -t from CLI. That script was provided as a hint. You actually don’t need it.

btx000, 2019-05-05:
I used a simple way to get kflash out of dependencies, which can now be run directly.

This may not be a perfect method.
[image]

```json
        {
            "sha1": "87ef5303917ff2e1fcc7b2bc194f67bd14e5f597",
            "system": "*",
            "url": "http://dl.sipeed.com/MAIX/platformio/dl-packages/tool-kflash-kendryte210-v0.8.2.tar.gz",
            "version": "0.8.2"
        },
```

ivankravets, 2019-05-05:
@btx000 Thanks! I think we can live with it :)

I've just deployed this package. I've just thought, we can actually can run `$PYTHONEXE -m pip install platform.get_package_dir("tool-kflash-kendryte210")` before uploading. In this case, we will install kflash dependencies to PIO Core site-packages.

Nevertheless, keeping packages in an isolated environment is a better option.

closed 2019-05-05 by ivankravets