sipeed/platform-kendryte210 #9: Debugging configuration

ivankravets, 2019-06-13:
I see you added debugging configuration for all boards. Do they really have FTDI + JLink on-board debugging probes?

carlosedp, 2019-06-19:
I have a similar problem on my Sipeed MaixGo board, while trying to use it with the Kendryte standalone SDK, I can't debug using it's internal link to see registers and instructions.

btx000, 2019-07-05:
Maix-go's on-board debugger open-ec is not yet developed, and can only be debugged using external ftdi or jlink.

ivankravets, 2019-07-09:
@btx000 which boards from this list https://github.com/sipeed/platform-kendryte210/tree/master/boards have on-board debugger(probe) and which are ready for external via JTAG?

closed 2019-07-11 by ivankravets

btx000, 2019-07-12:
> @btx000 which boards from this list https://github.com/sipeed/platform-kendryte210/tree/master/boards have on-board debugger(probe) and which are ready for external via JTAG?

All boards can be debugged using external JTAG，only maix go has an on-board debugger.

commit 755e27c7 (2019-07-11): Debugging support // Resolve #9, Resolve #10