sipeed/platform-kendryte210 #10: Debugging MaixGo on PlatformIO using Kendryte standalone SDK

carlosedp, 2019-06-19:
I have a problem on my Sipeed MaixGo board, while trying to use it with the Kendryte standalone SDK, I can't debug using it's internal link to see registers and instructions.

My platformio.ini is this:

```ini
[env:sipeed-maix-go]
platform = kendryte210
framework = kendryte-standalone-sdk
board = sipeed-maix-go
monitor_speed = 115200
debug_tool = jlink
```

Here is the error I get first:

```
Reading symbols from /Users/cdepaula/repos/maix-go/kendryte-blink/.pioenvs/sipeed-maix-go/firmware.elf...
done.
PlatformIO Unified Debugger > http://bit.ly/pio-debug
PlatformIO: Initializing remote target...
 _  __              _            _
| |/ /___ _ __   __| |_ __ _   _| |_ ___
| ' // _ \ '_ \ / _` | '__| | | | __/ _ \
| . \  __/ | | | (_| | |  | |_| | ||  __/
|_|\_\___|_| |_|\__,_|_|   \__, |\__\___|
                           |___/
Kendryte Open On-Chip Debugger For RISC-V v0.2.2 (2019-01-17)
Licensed under GNU GPL v2
embedded:startup.tcl:60: Error: Can't find kendryte_jlink.cfg
in procedure 'script'
at file "embedded:startup.tcl", line 60
.pioinit:10: Error in sourced command file:
Remote connection closed
```

Then, I changed `sipeed-maix-go.json` to have the absolute path for the cfg and get this error:

```
Reading symbols from /Users/cdepaula/repos/maix-go/kendryte-blink/.pioenvs/sipeed-maix-go/firmware.elf...
done.
PlatformIO Unified Debugger > http://bit.ly/pio-debug
PlatformIO: Initializing remote target...
 _  __              _            _
| |/ /___ _ __   __| |_ __ _   _| |_ ___
| ' // _ \ '_ \ / _` | '__| | | | __/ _ \
| . \  __/ | | | (_| | |  | |_| | ||  __/
|_|\_\___|_| |_|\__,_|_|   \__, |\__\___|
                           |___/
Kendryte Open On-Chip Debugger For RISC-V v0.2.2 (2019-01-17)
Licensed under GNU GPL v2
adapter speed: 3000 kHz
Error: No J-Link device found.

.pioinit:10: Error in sourced command file:
Remote connection closed

```

If I try to debug over FTDI using the kendryte_ftdi.cfg I got on the OpenOCD:

```
interface ftdi

ftdi_vid_pid 0x0403 0x6010

#func   io  dir val
#TCK    0   1   0
#TDI    1   1   0
#TDO    2   1   0
#TMS    3   0   1
#RST    4   1   1

#       val dir
#ftdi_layout_init 0x0028 0x002b
#只有jtagio分配
ftdi_layout_init 0x0008 0x000b

#ftdi_layout_init 0x0808 0x0a1b
#ftdi_layout_signal nSRST -oe 0x0200
#ftdi_layout_signal nTRST -data 0x0100 -oe 0x0100
#ftdi_layout_signal LED -data 0x0800

adapter_khz 500

gdb_port 3333
telnet_port 4444

set _CHIPNAME riscv

jtag newtap $_CHIPNAME cpu -irlen 5 -expected-id 0x04e4796b

set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME riscv -chain-position $_TARGETNAME

if {[ info exists pulse_srst]} {
  ftdi_set_signal nSRST 0
  ftdi_set_signal nSRST z
}

init
halt
```

I get this error:

```
Reading symbols from /Users/cdepaula/repos/maix-go/kendryte-blink/.pioenvs/sipeed-maix-go/firmware.elf...
done.
PlatformIO Unified Debugger > http://bit.ly/pio-debug
PlatformIO: Initializing remote target...
 _  __              _            _
| |/ /___ _ __   __| |_ __ _   _| |_ ___
| ' // _ \ '_ \ / _` | '__| | | | __/ _ \
| . \  __/ | | | (_| | |  | |_| | ||  __/
|_|\_\___|_| |_|\__,_|_|   \__, |\__\___|
                           |___/
Kendryte Open On-Chip Debugger For RISC-V v0.2.2 (2019-01-17)
Licensed under GNU GPL v2
adapter speed: 500 kHz
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : clock speed 500 kHz
Info : TAP riscv.cpu does not have valid IDCODE (idcode=0xfa82fa)
Info : JTAG tap: auto0.tap tap/device found: 0x007d417d (mfg: 0x0be (Seiko Epson), part: 0x07d4, ver: 0x0)
Info : JTAG tap: auto1.tap tap/device found: 0x007d007d (mfg: 0x03e (Oracle Corporation), part: 0x07d0, ver: 0x0)
Info : JTAG tap: auto2.tap tap/device found: 0x3cb5807d (mfg: 0x03e (Oracle Corporation), part: 0xcb58, ver: 0x3)
Info : TAP auto3.tap does not have valid IDCODE (idcode=0xffff8272)
Info : JTAG tap: auto4.tap tap/device found: 0xffffc139 (mfg: 0x09c (Eon Silicon Devices), part: 0xfffc, ver: 0xf)
Warn : AUTO auto0.tap - use "jtag newtap auto0 tap -irlen 2 -expected-id 0x007d417d"
Error: auto0.tap: IR capture error; saw 0x0003 not 0x0001
Warn : Bypassing JTAG setup events due to errors
Error: dtmcontrol is 0. Check JTAG connectivity/board power.
Info : Listening on port 3333 for gdb connections
Error: Target not examined yet

.pioinit:10: Error in sourced command file:
Remote connection closed
```

closed 2019-07-11 by ivankravets

ivankravets, 2019-07-11:
1. Please switch to the latest upstream version http://docs.platformio.org/en/latest/platforms/kendryte210.html#upstream
2. You need to configure [debug_tool](http://docs.platformio.org/en/latest/projectconf/section_env_debug.html#debug-tool). See currently supported https://github.com/sipeed/platform-kendryte210/blob/master/platform.py#L42

For example,
```
[env:sipeed-maix-go]
platform = kendryte210
framework = kendryte-standalone-sdk
board = sipeed-maix-go
monitor_speed = 115200
debug_tool = jlink
build_type = debug
```

See a demo from my `sipeed-maix-bit`
[image]

ivankravets, 2019-07-11:
@btx000, @GitJer , please re-test.

There are some issues with Kendryte toolchain where it returns a fake list of CSR* registers. We reported this to RISC-V official team and they fixed this bug a lot of months ago. It seems that @kendryte uses very old fork of RISC-V GCC. Ping @vowstar, @sunnycase. Can we use official RISC-V GCC toolchain?

ivankravets, 2019-07-11:
Please note that uploading through "upload_protocol" currently does not work. See issue https://github.com/kendryte/openocd-kendryte/issues/11

carlosedp, 2019-07-11:
But you are using an external debugger right? (I see Olimex on the config).

Can't I use the internal debugger from the MaixGo board?

ivankravets, 2019-07-11:
Does MaixGo have an internal debugger? I didn't see it.

carlosedp, 2019-07-11:
According to the docs, yes: http://docs.platformio.org/en/latest/boards/kendryte210/sipeed-maix-go.html#debugging

But according to the comment it's still not implemented or working: https://github.com/sipeed/platform-kendryte210/issues/9#issuecomment-508750193

ivankravets, 2019-07-11:
This is old and invalid docs. It was a mistake by @btx000. We will release a new version soon and re-generate docs.

@btx000 am I correct? MaixGo does not have on-board debugger?

btx000, 2019-07-12:
@ivankravets  Maix go board has an on-ec firmware based onboard debugger (using stm32 emulation ftdi2232), but the current version is not perfect, we are working for it.

ivankravets, 2019-07-12:
@btx000 please ping me when it will be supported.

adigandhi, 2019-07-16:
@Posting here to track this bug, I am unable to debug as well. Please keep us posted whenever we can debug this.

hiviah, 2020-01-28:
I had success with Flyswatter rev B to debug, which is another FTDI FT2232C based board. (Two JLinks I have didn't work after many attempts.) 

Single step, breakpoints, dump memory, and similar core commands seem to work now.

Command (openocd-kendryte revision `08bc16c3057be19927d6d5784e31abc208242719`:)

    openocd-kendryte/bin/openocd -f kendryte-flyswatter.cfg -m 0

Config file `kendryte-flyswatter.cfg`:

    # debug adapter
    interface ftdi
    ftdi_device_desc "Flyswatter"
    ftdi_vid_pid 0x0403 0x6010
    
    ftdi_layout_init 0x0818 0x0cfb
    ftdi_layout_signal nTRST -data 0x0010
    ftdi_layout_signal nSRST -oe 0x0020
    ftdi_layout_signal LED -data 0x0c00
    
    transport select jtag
    adapter_khz 3000
    
    # server port
    gdb_port 3333
    telnet_port 4444
    
    # add cpu target
    set _CHIPNAME riscv
    
    jtag newtap $_CHIPNAME cpu -irlen 5 -expected-id 0x04e4796b
    
    set _TARGETNAME $_CHIPNAME.cpu
    target create $_TARGETNAME riscv -chain-position $_TARGETNAME
    
    # command
    init
    halt

Kabron287, 2020-10-10:
Hello, 
strange that I do not see mention about disabling onboard STM32 for external debug.
It is obvious that active STM32 interface will prevent successful debug via external probe.
And I do observe this collision when trying to connect J-link. J-link could not pull JTAG lines to ground.
So, there must be a way to disable STM32. Maybe by pulling reset down, but it is inconvenuet.

Kabron287, 2020-10-10:
I disabled STM32 via BOOT0 during PowerON. Now I see correct oscillograms with J-link.
My output looks like:
`Kendryte Open On-Chip Debugger For RISC-V v0.2.3 (2019-02-21)
Licensed under GNU GPL v2
debug_level: 2
adapter speed: 1000 kHz
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
riscv.cpu
Info : J-Link V9 compiled Sep  1 2016 18:29:50
Info : Hardware version: 9.20
Info : VTarget = 3.269 V
Info : clock speed 1000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x04e4796b (mfg: 0x4b5 (<unknown>), part: 0x4e47, ver: 0x0)
Core [0] halted at 0x404 due to software breakpoint
Info : Examined RISCV core; found 2 harts
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: riscv.cpu tap/device found: 0x04e4796b (mfg: 0x4b5 (<unknown>), part: 0x4e47, ver: 0x0)
** Programming Started **
embedded:startup.tcl:476: Error: ** Programming Failed **
in procedure 'program' 
in procedure 'program_error' called at file "embedded:startup.tcl", line 532
at file "embedded:startup.tcl", line 476
*** [upload] Error 1`
Segger Jlink says that my J-link does not support this RISKV debug via JTAG. Could it be a reason?

commit 755e27c7 (2019-07-11): Debugging support // Resolve #9, Resolve #10