sipeed/platform-kendryte210 #12: Please update kflash and maixduino to the new version.

btx000, 2019-07-05:
```
        {
            "sha1": "e35862cf42b83c71c0be1a38d9b5857b6d94cf3e",
            "system": "*",
            "url": "http://dl.sipeed.com/MAIX/platformio/dl-packages/framework-maixduino-0.3.9.tar.gz",
            "version": "0.3.9"
        },
        {
            "sha1": "e559a9d6655ae08990ed846c760c68855b43e2e6",
            "system": "*",
            "url": "http://dl.sipeed.com/MAIX/platformio/dl-packages/tool-kflash-kendryte210-v0.9.1.tar.gz",
            "version": "0.9.1"
        },
```

ivankravets, 2019-07-05:
Thanks! Done!

closed 2019-07-05 by ivankravets

dimitre, 2020-10-24:
For some reason kflash from platformio is really unstable here
0.9.1 is always stuck in 
```
[WARN] Timeout Error, retrying...
```
so I compile in platformio and upload via command line with kflash v. 0.8.3

I'm using Macos 10.14.6 and Maix Bit with mic board.