sipeed/platform-kendryte210 #16: Update kendryte-standalone-sdk to v0.5.6

btx000, 2019-07-18:
{
            "sha1": "8ba94f5ff87eec5e13e488d98050b5e53fab2125",
            "system": "*",
            "url": "http://dl.sipeed.com/MAIX/platformio/dl-packages/framework-kendryte-standalone-sdk-0.5.6.tar.gz",
            "version": "0.5.6"
        }，

ivankravets, 2019-07-18:
Done! Do you plan to do any changes this week? Could we make the next release?

closed 2019-07-18 by ivankravets

btx000, 2019-07-18:
No other changes.

ivankravets, 2019-07-18:
@btx000 is it possible to PlatformIO listed here https://maixduino.sipeed.com/en/? As an alternative to Arduino IDE with screenshots and debugging. I'm sure developers will like it.

btx000, 2019-07-18:
Of course, I will add.

ivankravets, 2019-07-18:
Thanks! v1.1.1 is out! https://github.com/sipeed/platform-kendryte210/releases/tag/v1.1.1