sipeed/platform-kendryte210 #17: Cannot Blink Built-in LED with Basic Example Code

baqwas, 2019-07-29:
- Desktop: **Ubuntu 19.04**
- IDE: **Arduino 1.8.9**
- Kendryte K210 library: **?**
- Board: **Sipeed Maixduino**

Upload displays _Done Saving_, on-board LED blinks during process but **no steady blink that occurs when code is running**. Still cannot blink built-in LED (pin # 13?). Is there a checklist that I can follow to ensure that I have not omitted any step?

Thanks.

Output pane is _hung_ with:

Downloading ISP: |```
=============================================================================================================================================================================================================================================================================================================================================================================================================================================================| 100.0% 62kiB/s
[INFO] Booting From 0x80000000 
[INFO] Wait For 0.1 second for ISP to Boot 
[INFO] Boot to Flashmode Successfully 
[INFO] Selected Baudrate:  1500000 
[INFO] Baudrate changed, greeting with ISP again ...  
[INFO] Boot to Flashmode Successfully 
[INFO] Selected Flash:  On-Board 
[INFO] Initialization flash Successfully 

Programming BIN: |==============================================================================================================================================================================================================================-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------| 50.0% 52kiB/s

Programming BIN: |=============================================================================================================================================================================================================================================================================================================================================================================================================================================================| 100.0% 54kiB/s
[INFO] Rebooting... 
```
There is nothing visible occurring after the last line is posted in the Output pane.

There should be some documentation _somewhere_ but somehow owing to my inexperience I don't know where to find it. Please help me. I would like to test the Sipeed Maixduino board with beginner's examples.

Kind regards.

feldim2425, 2019-08-31:
Try chaning the Baudrate to: 115200

baqwas, 2019-08-31:
@feldim2425 Thx for the suggestion. _Been_ there, **done** that before posting my issue. I have suspended the Arduino effort and will try the MicroPython approach given that @loboris is guiding me.
Kind regards.