sipeed/platform-kendryte210 #18: Problems running the debugger

feldim2425, 2019-08-28:
I cannot use the sipeed-rv-debugger to debug my MaixDock.
It outputs the following in the debug console: 
```
 _  __              _            _     
| |/ /___ _ __   __| |_ __ _   _| |_ ___  
| ' // _ \ '_ \ / _` | '__| | | | __/ _ \ 
| . \  __/ | | | (_| | |  | |_| | ||  __/ 
|_|\_\___|_| |_|\__,_|_|   \__, |\__\___| 
                           |___/          
Kendryte Open On-Chip Debugger For RISC-V v0.2.3 (2019-02-21)
Licensed under GNU GPL v2
ftdi samples TDO on falling edge of TCK
adapter speed: 1000 kHz
riscv.cpu
Info : tcl server disabled
Info : telnet server disabled
Info : clock speed 1000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0xfe030713 (mfg: 0x389 (Lantiq Deutschland GmbH), part: 0xe030, ver: 0xf)
Warn : JTAG tap: riscv.cpu       UNEXPECTED: 0xfe030713 (mfg: 0x389 (Lantiq Deutschland GmbH), part: 0xe030, ver: 0xf)
Error: JTAG tap: riscv.cpu  expected 1 of 1: 0x04e4796b (mfg: 0x4b5 (<unknown>), part: 0x4e47, ver: 0x0)
Error: Trying to use configured scan chain anyway...
Warn : Bypassing JTAG setup events due to errors
Error: dtmcontrol is 0. Check JTAG connectivity/board power.
Info : accepting 'gdb' connection from pipe
openocd: src/target/riscv/riscv.c:1496: riscv_xlen_of_hart: Assertion `r->xlen[hartid] != -1' failed.
Aborted (core dumped)
.pioinit:13: Error in sourced command file:
Remote connection closed
```

I checked the connection multiple times they seem to be ok.
platform.ini :
```
[env:sipeed-maix-one-dock]
platform = https://github.com/sipeed/platform-kendryte210.git
board = sipeed-maix-one-dock
framework = kendryte-freertos-sdk
upload_protocol = kflash
debug_tool = sipeed-rv-debugger
```

Operating System: Ubuntu 18.10

feldim2425, 2019-08-29:
btw:
Connections: 
``
Sipeed Risc-V debugger -> Maix M1w Dock
Pin 1 (TCK) -> Pin 0 (TCK) [Next to the USB Plug]
Pin 2 (GND) -> GND [Top Connector, Above the RST Pin]
Pin 3 (TDO) -> Pin 3 (TDO) [Next to the USB Plug]
Pin 5 (TMS) -> Pin 2 (TMS) [Next to the USB Plug]
Pin 7 (TMS) -> RST [Top Connector]
Pin 9 (TDI) -> Pin 1 (TDI) [Next to the USB Plug]
``

jamesgraves, 2020-08-30:
For anyone else seeing this issue, this is definitely a JTAG cable problem. Re-check all pins for continuity and short circuits.

Kabron287, 2020-10-10:
Hello,
strange that I do not see mention about disabling onboard STM32 for external debug.
It is obvious that active STM32 interface will prevent successful debug via external probe.
And I do observe this collision when trying to connect J-link. J-link could not pull JTAG lines to ground.
So, there must be a way to disable STM32. Maybe by pulling reset down, but it is inconvenuet.

I disabled STM32 via BOOT0 during PowerON. Now I see correct oscillograms with J-link.

My output looks like:
Kendryte Open On-Chip Debugger For RISC-V v0.2.3 (2019-02-21) Licensed under GNU GPL v2 debug_level: 2 adapter speed: 1000 kHz Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'. riscv.cpu Info : J-Link V9 compiled Sep 1 2016 18:29:50 Info : Hardware version: 9.20 Info : VTarget = 3.269 V Info : clock speed 1000 kHz Info : JTAG tap: riscv.cpu tap/device found: 0x04e4796b (mfg: 0x4b5 (<unknown>), part: 0x4e47, ver: 0x0) Core [0] halted at 0x404 due to software breakpoint Info : Examined RISCV core; found 2 harts Info : Listening on port 3333 for gdb connections Info : JTAG tap: riscv.cpu tap/device found: 0x04e4796b (mfg: 0x4b5 (<unknown>), part: 0x4e47, ver: 0x0) ** Programming Started ** embedded:startup.tcl:476: Error: ** Programming Failed ** in procedure 'program' in procedure 'program_error' called at file "embedded:startup.tcl", line 532 at file "embedded:startup.tcl", line 476 *** [upload] Error 1
Segger Jlink says that my J-link does not support this RISKV debug via JTAG. Could it be a reason?
PS
If I do not use upload via J-link, debug starts and works correctly. So Segger is no right.

Kabron287, 2020-10-10:
To my surprise, if I do not declare **upload_protocol** everything works fine

vincentabraham, 2021-09-09:
I'm having the same problem with debugging using the Nexys A7 50T. Please have a look at the .doc file attached with this post and help me resolve this issue,
[Uploading nexys50t debug.docx…]()

talpachen, 2022-03-27:
K210 TDO pin driving ability is too weak, not support some adapter without Isolated chip.

First test attached logic analyzer:
```
> .\openocd.exe -f .\jlink_kendryte.cfg
xPack OpenOCD x86_64 Open On-Chip Debugger 0.11.0+dev (2021-12-07-17:33)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter driver' not 'interface'
DEPRECATED! use 'adapter speed' not 'adapter_khz'
Info : J-Link ARM V8 compiled Nov 28 2014 13:44:46
Info : Hardware version: 8.00
Info : VTarget = 3.300 V
Info : clock speed 1000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0xfe018389 (mfg: 0x1c4 (Satron Handelsges), part: 0xe018, ver: 0xf)
Warn : JTAG tap: riscv.cpu       UNEXPECTED: 0xfe018389 (mfg: 0x1c4 (Satron Handelsges), part: 0xe018, ver: 0xf)
Error: JTAG tap: riscv.cpu  expected 1 of 1: 0x04e4796b (mfg: 0x4b5 (Canaan-Creative Co Ltd), part: 0x4e47, ver: 0x0)
Error: Trying to use configured scan chain anyway...
Warn : Bypassing JTAG setup events due to errors
Error: dtmcontrol is 0. Check JTAG connectivity/board power.
Warn : target riscv.cpu examination failed
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
Error: Target not examined yet
```

Second test:
```
> .\openocd.exe -f .\jlink_kendryte.cfg
xPack OpenOCD x86_64 Open On-Chip Debugger 0.11.0+dev (2021-12-07-17:33)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter driver' not 'interface'
DEPRECATED! use 'adapter speed' not 'adapter_khz'
Info : J-Link ARM V8 compiled Nov 28 2014 13:44:46
Info : Hardware version: 8.00
Info : VTarget = 3.300 V
Info : clock speed 1000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x04e4796b (mfg: 0x4b5 (Canaan-Creative Co Ltd), part: 0x4e47, ver: 0x0)
Info : [riscv.cpu] Found 4 triggers
halted at 0x800b5256 due to debug interrupt
Info : Examined RISCV core; XLEN=64, misa=0x800000000014112d
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
```

spg-one, 2022-11-04:
> K210 TDO pin driving ability is too weak, not support some adapter without Isolated chip.
>
> First test attached logic analyzer:
>
> ```
> xPack OpenOCD x86_64 Open On-Chip Debugger 0.11.0+dev [...]
can you tell more details about how you made this？thanks

spg-one, 2022-11-04:
im using the rv debugger lite , and facing excatly the same problem

chunyexixiaoyu, 2022-12-13:
i have the same problem，using jlink ultra+ can not connect the k210 module。but can connect sipeed MAXbit
(with chip)
`PS E:\kendryte-openocd-0.2.3-win32\kendryte-openocd> .\bin\openocd.exe -f .\tcl\kendryte.cfg
 _  __              _            _
| |/ /___ _ __   __| |_ __ _   _| |_ ___
| ' // _ \ '_ \ / _` | '__| | | | __/ _ \
| . \  __/ | | | (_| | |  | |_| | ||  __/
|_|\_\___|_| |_|\__,_|_|   \__, |\__\___|
                           |___/
Kendryte Open On-Chip Debugger For RISC-V v0.2.3 (2019-02-21)
Licensed under GNU GPL v2
adapter speed: 500 kHz
Info : J-Link Ultra V4 compiled Sep 22 2022 15:00:10
Info : Hardware version: 4.00
Info : VTarget = 3.305 V
Info : clock speed 500 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: riscv.cpu: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: dtmcontrol is 0. Check JTAG connectivity/board power.
Info : Listening on port 3333 for gdb connections
Error: Target not examined yet`

chunyexixiaoyu, 2022-12-13:
[image]