sipeed/platform-kendryte210 #19: Board: sipeed-maixduino,  framework=arduino

fivefishstudios, 2019-09-21:
Platformio Build works fine. 
But I'm getting this error when trying to Upload to a Maxiduino board. 

File "/Users/username/.platformio/packages/tool-kflash-kendryte210/kflash.py", line 13, in <module>
    import argparse
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/argparse.py", line 87, in <module>
    import re as _re
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/re.py", line 143, in <module>
    class RegexFlag(enum.IntFlag):
AttributeError: module 'enum' has no attribute 'IntFlag'
*** [upload] Error 1

This enum error only seems to affect when uploading to Sipeed/Kendryte boards. 

PIO uploading to ESP8266 and ESP32 boards work fine.  Only when uploading to Kendryte platform do I get this error.

btx000, 2019-09-25:
Hi, @fivefishstudios 

In order to make kflash run in python2, the enum module is placed in the kflash directory, which will be encountered in recent python3. You can open the kflash directory like this
`C:\Users\{your user name}\.platformio\packages\tool-kflash-kendryte210`
then delete the `enum` folder and test it again and it should be able to download successfully.

fivefishstudios, 2019-09-25:
Thank you so much! It's working now.

closed 2019-09-25 by fivefishstudios

dimitre, 2020-01-28:
I'm having a similar issue here
```
autechre:arduino-blink z$ pio run
Processing sipeed-maix-bit (platform: kendryte210; framework: arduino; board: sipeed-maix-bit)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------
Error in sitecustomize; set PYTHONVERBOSE for traceback:
AttributeError: module 'enum' has no attribute 'IntFlag'
Traceback (most recent call last):
  File "/Users/z/.platformio/packages/tool-scons/script/scons", line 104, in <module>
    import pkg_resources
  File "/usr/local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 24, in <module>
    import re
  File "/usr/local/Cellar/python/3.7.6_1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/re.py", line 143, in <module>
    class RegexFlag(enum.IntFlag):
AttributeError: module 'enum' has no attribute 'IntFlag'
```

And I didn't find this enum file to move or rename. any ideas?
[image]

edit: now I can build everything after removing enum34 using command
```
sudo -H pip uninstall enum34
```

valeros, 2020-01-28:
Hi @dimitre ! This issue should have been fixed already. Could you please upload to https://hastebin.com/ the contents of `/Users/z/.platformio/packages/tool-scons/tool-kflash-kendryte210/kflash.py` file?
Thanks!

dimitre, 2020-01-28:
Now it is working here but this file is not present
```
ls -alF /Users/z/.platformio/packages/tool-scons/
total 1720
drwx------  16 z  staff     512 Jan 28 01:17 ./
drwxr-xr-x  12 z  staff     384 Jan 28 03:18 ../
-rwxr-xr-x   1 z  staff  238440 Aug  8 11:50 CHANGES.txt*
-rw-r--r--   1 z  staff    1071 Aug  8 11:50 LICENSE.txt
-rw-r--r--   1 z  staff    5795 Aug  8 11:50 MANIFEST
-rw-r--r--   1 z  staff     451 Aug  8 11:50 PKG-INFO
-rw-r--r--   1 z  staff    8266 Aug  8 11:50 README.txt
-rwxr-xr-x   1 z  staff    2620 Aug  8 11:50 RELEASE.txt*
drwxr-xr-x   3 z  staff      96 Jan 28 01:17 engine/
-rwxr-x---   1 z  staff     146 Aug 26 15:53 package.json*
-rw-r--r--   1 z  staff   26025 Aug  8 11:50 scons-time.1
-rw-r--r--   1 z  staff  535346 Aug  8 11:50 scons.1
-rw-r--r--   1 z  staff    5226 Aug  8 11:50 sconsign.1
drwxr-xr-x   7 z  staff     224 Jan 28 01:17 script/
-rw-r--r--   1 z  staff     127 Aug  8 11:50 setup.cfg
-rwxr-xr-x   1 z  staff   21793 Aug  8 11:50 setup.py*
```

valeros, 2020-01-28:
@dimitre Sorry, that's my bad, I gave you a wrong path. Here is the correct one `/Users/z/.platformio/packages/tool-kflash-kendryte210/kflash.py`

dimitre, 2020-01-28:
here
https://hastebin.com/gezacocupi.py
I removed all kflash installed through pip and pip3 and platformio installed it automagically after removing enum34