sipeed/platform-kendryte210 #26: Support for ARM CPUs

GiorgioGhisotti, 2020-07-18:
There is no ARM support for the K210 toolchain in the PlatformIO repository; to make this work on an ARM device (e.g. a raspberry pi) you need to compile the [toolchain](https://github.com/kendryte/kendryte-gnu-toolchain) from github and remove the entry from platform.json in ~/.platformio/platforms/kendryte210:

before:
```json
 "packages": {
     "toolchain-kendryte210": {
       "type": "toolchain",
       "version": "~8.2.0"
     },
     "framework-maixduino": {
       "type": "framework",
       "optional": true,
       "version": "~0.3.9"
     },
```

after:
```json
 "packages":
     "framework-maixduino": {
       "type": "framework",
       "optional": true,
       "version": "~0.3.9"
     },
```

It might be a good idea to specify this in the README or ask PlatformIO to add an ARM release (the toolchain builds perfectly fine so this shouldn't be an issue).

ivankravets, 2020-07-19:
Could you provide a TAR.GZ archive of compiled toolchain on Arm machine? We will deploy it to the our registry.

GiorgioGhisotti, 2020-07-19:
There is an archive posted by the developers in [this](https://github.com/kendryte/kendryte-gnu-toolchain/issues/6#issuecomment-541622351) issue, it should be up to date since the source code hasn't changed since then.

ivankravets, 2020-07-19:
The package has been published in the PlatformIO Regsitry. Could you try to build project again without custom toolchain? Does it work now?

GiorgioGhisotti, 2020-07-19:
I restored the old platform.json file and it works perfectly, problem solved :) thank you for acting so quickly!

closed 2020-07-19 by GiorgioGhisotti

ivankravets, 2020-07-19:
Happy coding with PlatformIO! ;)

ivankravets, 2020-07-19:
Happy coding with PlatformIO! ;)

P.S: @loboris, thanks for the package!