sipeed/platform-kendryte210 #37: Sipeed Maix Amigo board support

kallisti5, 2020-10-29:
Seeed Studio is now shipping the Sipeed Maix Amigo devices. These are powered via the Kendryte K210 (like the other Maix devices)

### Platform information

* SoC – Kendryte K210 Dual-core 64-bit RISC-V (RV64GC) processor with FPU @ 400 MHz (overclockable to 500MHz), 8MB SRAM, built-in AI accelerators for video and audio
* Storage – 16MB Flash, MicroSD card slot up
* Display – 3.5-inch TFT capacitive touch screen display with 480×320 resolution
* Camera
    * VGA front-facing camera up to 30 fps (GC0328 sensor)
    * VGA rear camera up to 60 fps (OV7740 sensor)
* Audio – 6+1 microphone array
* USB – 1x USB Type-C port for power and programming
* Expansion
    * 3x Grove ports with I2C, GPIO, etc…
    * 3x 8-pin SP-MOD headers with GPIOs, 3.3V, and GND signals; compatible with ESP-01 ESP8266 WiFi module, PSRAM modules, etc…
* Sensor – Accelerometer
* Misc – Reset button, 3x function buttons, 3x LEDs
* Battery – 520 mAh Lithium battery
* Power Supply – Via USB-C port
* Dimensions – 104.3 x 63.3 x 16.5mm

It looks like Sipeed has done "something" here for platformio. Guessing it wasn't upstreamed
https://dl.sipeed.com/MAIX/platformio/dl-packages

kallisti5, 2020-10-30:
Looks like they have a library externally:
https://github.com/sipeed/platform-kendryte210

kallisti5, 2020-11-28:
@ivankravets  can you help here?  Seeedstudio documents PlatformIO support on the Maix Amigo, but there is zero support.
It doesn't look like a *lot* of work needs done, but the board needs defined within platform-kendryte210

junhuanchen, 2020-11-29:
Will. This needs to be added to the configuration file, but not yet processed.

vamoosebbf, 2021-02-21:
> @ivankravets can you help here? Seeedstudio documents PlatformIO support on the Maix Amigo, but there is zero support.
> It doesn't look like a _lot_ of work needs done, but the board needs defined within platform-kendryte210

[image]
Both MaixCube and MaixAmigo can choose to develop MaixDuino, they are compatible, refer to the configuration above