sipeed/platform-kendryte210 #38: MAIXCube Support

kisdia, 2020-11-10:
Sipeed MaixCube – All-in-One AI Development Platform Based on K210 (RISC-V) now available. https://www.seeedstudio.com/Sipeed-Maix-Cube-p-4553.html

New platform needs to be added.

kallisti5, 2020-11-28:
@ivankravets can you help here? Seeedstudio documents PlatformIO support on the MAIXCube, but there is zero support here.

It doesn't look like a lot of work needs done, but the board needs defined within platform-kendryte210

vamoosebbf, 2021-02-21:
[image]
Both MaixCube and MaixAmigo can choose to develop MaixDuino, they are compatible, refer to the configuration above

vamoosebbf, 2021-02-21:
> @ivankravets can you help here? Seeedstudio documents PlatformIO support on the MAIXCube, but there is zero support here.
> 
> It doesn't look like a lot of work needs done, but the board needs defined within platform-kendryte210

[image]
Both MaixCube and MaixAmigo can choose to develop MaixDuino, they are compatible, refer to the configuration above