sipeed/platform-kendryte210 #40: tool-openocd-kendryte not available for linux_armv7l

maxgerhardt, 2021-04-05:
Per https://community.platformio.org/t/platformio-tool-openocd-kendryte-for-system-linux-armv7l/20420. 

When using a Raspberry Pi with Raspbian GNU/Linux 10 (buster) as OS installed (32-bit), one cannot upload to the boards because `tool-openocd-kendryte` was not uploaded for this configuration.

```
Processing sipeed-maix-one-dock (platform: kendryte210; board: sipeed-maix-one-dock; framework: kendryte-standalone-sdk)
--------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing platformio/tool-openocd-kendryte @ ^1.203.0
Error: Could not find the package with 'platformio/tool-openocd-kendryte @ ^1.203.0' requirements for your system 'linux_armv7l'
The terminal process "platformio 'run', '--target', 'upload', '--environment', 'sipeed-maix-one-dock'" terminated with exit code: 1.
```

Looking at https://api.registry.platformio.org/v3/packages/platformio/tool/tool-openocd-kendryte, it's only available for x86_64 Windows, Linux and Mac, but not for the ARM32 and ARM64 types.

maxgerhardt, 2021-04-05:
A precompiled version is available at https://github.com/xalogic-open/tool-openocd-kendryte/releases/download/v1.203.1_linux_armv7l/tool-openocd-kendryte-linux_armv7l-1.203.1.tar.gz and can be included in the `platformio.ini` in other projects per 

```
lib_deps = 
   tool-openocd-kendryte@https://github.com/xalogic-open/tool-openocd-kendryte/releases/download/v1.203.1_linux_armv7l/tool-openocd-kendryte-linux_armv7l-1.203.1.tar.gz
```
([docs](https://docs.platformio.org/en/latest/projectconf/section_env_platform.html#platform-packages))

ivankravets, 2021-04-05:
@maxgerhardt  thanks! The package has been published to the registry.

closed 2021-04-05 by ivankravets